rrivirr/rriv-ctl #20: rrivctl feature flags

ZavenArra, 2025-09-15:
Any rrivctl command not explicitly releases as stable must be hidden behind a feature flag, which can then be enabled by adding a corresponding environment variable to the .env file.

For instance, the config-history feature is not released yet as stable.  So this must be enabled by the user by adding the following .env
`FEATURE_CONFIG_HISTORY=true`

Kpoke, 2025-09-16:
Given that we are using binary builds of the cli, I'm not sure this is the way to go with this. I think anyone looking to use such features would make use of builds from the alpha/beta branches

ZavenArra, 2025-11-11:
We don't need this until we force everyone onto rrivctlv2.

ZavenArra, 2026-01-20:
won't build.