rrivirr/rriv-ctl #33: launch rrivctl with default context

ZavenArra, 2025-09-16:
The user should not have to create a context manually the first time they run rrivctl, we must create a default context for them.  This is because they will not be familiar with the concept of a context.

When using the default context, the prompt will simply be 
`rrivctl> `

Kpoke, 2025-09-16:
Do we want to do that? I think, If we have a default context and users start working in the default context but then they eventually understand the idea of contexts. They would have to move their devices to the right contexts as they'd prefer which could be tedious. Also, that could introduce a 'laziness to fully understand and use contexts'

ZavenArra, 2025-09-16:
i think that's ok. in a lot of ways the context concept is an advanced one for our users.   some of our users will immediately understand the concept and use it right away, and have no problem moving devices between contexts.  others will find it confusing and it will be a constant point of friction and support needs.   most users will have a single device for the first 3-6 months of using the platform, so they won't need contexts until they are deploying or doing more with the system.  

so yes, I think we do want a default context.   but we can think about how we introduce contexts to the user, like maybe when they add their 2nd or 3rd device that's the time to have some text 'Adding your second device?  It might be a good time to learn about the context feature.  More details at this link'.  Something like that.